contrem/arduino-timer #1: Function return value interpretation

Floessie, 2018-10-10:
Hi Michael,

This is one of the better timer libraries for Arduino I've seen so far. :+1: It does without dynamic allocation and makes use of `const`. I hope you've tested the timer overflow. :wink: 

Only a personal nitpicking: I would change the meaning of the `function_to_call()` return value. `false` is negative, intuitively meaning "I don't want no more." I would have expected it to *not* be called again when returning a `false`.

Just my 2¢,
Flössie

contrem, 2018-10-11:
Thanks, Flössie.

Timer rollover / overflow is tested, you can check out the tests in the [extras/tests/rollover](/contrem/arduino-timer/tree/ff363ba52673a9c883bf2f52dae6f777558b34ec/extras/tests/rollover) and [extras/tests/rollover-generic](/contrem/arduino-timer/tree/ff363ba52673a9c883bf2f52dae6f777558b34ec/extras/tests/rollover-generic) sketches.

As for the return value, my interpretation is "remove the timer? false," but I can see your point. Since it's already published, I'm going to leave it as is unless there's more feedback requesting otherwise.

I appreciate the feedback.

Floessie, 2018-10-12:
@contrem 
> Timer rollover / overflow is tested, you can check out the tests in the extras/tests/rollover and extras/tests/rollover-generic sketches.

Well done! :+1: I'm impressed. And I :heart: lambdas.

> As for the return value, my interpretation is "remove the timer? false," but I can see your point. Since it's already published, I'm going to leave it as is unless there's more feedback requesting otherwise.

I see, you made up your mind. Much appreciated!

While reading the code, I came up with another list of random thoughts. This doesn't mean to push you in any way, just brainstorming:

- As this is C++11, you could turn the `typedef bool (*handler_t)(void *opaque)` into a `using handler_t = ...`. 
- Thinking about C++11, why not use `nullptr`?
- Why is `tick(unsigned long t)` public? Doesn't that open a door for cheating?
- In `tick(unsigned long t)` you copy all members of the task. Wouldn't it save stack and binary size to use the members directly (optimizer aside)?
- I could imagine a `_TIMER_H__` define could clash with other timer libraries (not than one would use another library voluntarily knowing this one :wink:), so I'd propose a `#pragma once` which is basically standard amongst compilers.
- I share your addiction to `const`, but some pointers aren't constified consequently. E.g. `struct task *slot = ` could really be `struct task* const slot = ` in many places.
- If I'm not missing something, `remove()` could be turned into a [one-liner](https://en.cppreference.com/w/cpp/language/zero_initialization): `*task = {};`

Again, this is a very slim and sleek library. I can only recommend it.

Best,
Flössie

contrem, 2018-10-12:
> I see, you made up your mind. Much appreciated!

The more I think about it, the more it makes sense, so it's changed in the [next version](/contrem/arduino-timer/tree/next), which will be 1.0.0. [mc/func-return-value-interp](/contrem/arduino-timer/tree/mc/func-return-value-interp) will track the changes from this issue until it's merged to master. Feel free to review.

> * As this is C++11, you could turn the `typedef bool (*handler_t)(void *opaque)` into a `using handler_t = ...`.
> * Thinking about C++11, why not use `nullptr`?
> * If I'm not missing something, `remove()` could be turned into a [one-liner](https://en.cppreference.com/w/cpp/language/zero_initialization): `*task = {};`

The library is currently not c++11, but it's compatible with both c++98 and c++11. Great suggestions for if c++11 becomes the target.

> * In `tick(unsigned long t)` you copy all members of the task. Wouldn't it save stack and binary size to use the members directly (optimizer aside)?

With optimization, it changes when the memory is read. This way should take advantage of memory pref etching / caching, but most of these chips probably don't have any of that, so I'll change it to be clearer.

> * Why is `tick(unsigned long t)` public? Doesn't that open a door for cheating?

That's exactly what it's for.

> * I could imagine a `_TIMER_H__` define could clash with other timer libraries (not than one would use another library voluntarily knowing this one wink), so I'd propose a `#pragma once` which is basically standard amongst compilers.

Good call, I'll make it a more unique name.

> * I share your addiction to `const`, but some pointers aren't constified consequently. E.g. `struct task *slot =` could really be `struct task* const slot =` in many places.

I noticed this while doing the return value change. These will be changed in the next version.

Floessie, 2018-10-12:
> The library is currently not c++11, but it's compatible with both c++98 and c++11. Great suggestions for if c++11 becomes the target.

Ah, okay, I see. I wrongly extrapolated the examples to the lib. Sorry.

> With optimization, it changes when the memory is read. This way should take advantage of memory prefetching / caching, but most of these chips probably don't have any of that, so I'll change it to be clearer.

No, definitely not, speaking for AVR. It's not even von-Neumann. Saving flash and RAM should be the goal.

As I'm concerned with AVRs: You probably know they have several sleep modes, and I often use the watchdog as a tick source and go down to lowest power. Imagine, your `tick()` would return the number of ticks until the next event instead of a `void`. Then one could dynamically sleep. That's still a very rough idea, though.

Keep on the good work! :+1: 

Best,
Flössie

contrem, 2018-10-12:
> As I'm concerned with AVRs: You probably know they have several sleep modes, and I often use the watchdog as a tick source and go down to lowest power. Imagine, your `tick()` would return the number [...]

Nice idea. I think a separate function for that would be ideal. Please open a separate issue for this feature if you're interested.

commit a9b8cad1 (2018-10-12): Merge branch 'mc/func-return-value-interp' into next

* mc/func-return-value-interp:
  src/timer: more unique include guard
  src/timer: clean up tick loop
  src/timer: consistent use of const
  arduino-timer: handler return value interpretation

Closes: #1
closed 2018-10-12 by contrem