contrem/arduino-timer #3: cancel a timer before it gets executed

bogomips, 2019-05-05:
I think that having the possibility of canceling a task before it gets executed could be very useful.
Here is how I've implemented it https://github.com/bogomips/arduino-timer

contrem, 2019-05-05:
Nice job. I avoided adding cancels to not have to deal timer IDs and their issues (wrapping, invalid reuse, etc).

However, it is still possible to have a 'cancel-able' timer by passing a flag as the optional argument to the called function (in / at / every) that the callback can check before running. Something like (untested, but the concept is valid):

```cpp
bool active = true;

bool toggle_led(void *_active) {
  bool is_active = *(bool *)_active;

  if (!is_active) return false;

  digitalWrite(LED_BUILTIN, !digitalRead(LED_BUILTIN)); // toggle the LED
  return true; // keep timer active? true
}

void setup() {
  pinMode(LED_BUILTIN, OUTPUT); // set LED pin to OUTPUT

  // call the toggle_led function every 1000 millis (1 second)
  timer.every(1000, toggle_led, &active);
}

void loop() {
  timer.tick(); // tick the timer
  active = false; // cancel the timer
}
```

Either way, you have to maintain a list of flags or task id's to know which timer to cancel. I'm not opposed to adding this type of functionality into the library if it's a highly requested feature, but I think that the overhead of using task id's is pretty much the same as the flag approach detailed above.

Some issues I noticed with your implementation:

add_task returns NULL with the timer array is full in these locations:

- https://github.com/bogomips/arduino-timer/blob/master/src/timer.h#L63
- https://github.com/bogomips/arduino-timer/blob/master/src/timer.h#L73
- https://github.com/bogomips/arduino-timer/blob/master/src/timer.h#L82

And so will crash when you dereference _task to get the id without checking for NULL return:

```cpp
        _task = add_task(now, time - now, h, opaque);
        return _task->id; // crash here when full
```

Using a random ID, though very small probability, can still return duplicate IDs. An incrementing a counter, as long as you handle wrapping, would be better:
- https://github.com/bogomips/arduino-timer/blob/master/src/timer.h#L170

Let me know what you think.

bogomips, 2019-05-06:
Hi and thanks for your quick answer!

I thought about using a variable as an argument to the called function but I immediately realized that it works better for small portions of code. IMHO The resulting code is not elegant or very readable.
I come from Javascript and it was so natural to search for a cancel() 
Fortunately, your code is clear enough and has already the useful private method remove()

I have no idea if this feature is highly requested; for me, my background, the way my brain works, having a cancel() is extremely more intuitive/readable/straightforward.

Also, thanks for checking my code and sharing the issues you've noticed.
I wrote everything in a rush as I needed it to finish the application I was working on, that's why, for example, I preferred to use a random number rather than a counter.
I will surely spend some time more to fix these problems.

if I reach a "decent" point it would be nice if you would implement this feature on your code, of course, you can also modify whatever you like to match your coding style, naming convention or adding any kind of improvement/bugfix.

Any input will always be extremely appreciated.

bogomips, 2019-05-06:
Actually, fixing the crash should be as simple as
`       
  inline
    struct task *
    add_task(unsigned long start, unsigned long expires,
             handler_t h, void *opaque, bool repeat = 0)
    {
        struct task * const slot = next_task_slot();

        if (!slot) 
            slot->id = random(2147483646);
        else
            slot->id = 0;

        slot->handler = h;
        slot->opaque = opaque;
        slot->start = start;
        slot->expires = expires;
        slot->repeat = repeat;

        return slot;
    }
`

contrem, 2019-05-06:
There is still a crash in your suggested fix:
> Actually, fixing the crash should be as simple as
> 
> ```
>         if (!slot) 
>             slot->id = 0; // this will crash here, dereferencing a NULL slot
> ```

One fix would be to check the return of add_task() in the in / at / every functions before accessing the ID, and return an invalid ID if it's NULL.

In terms of inclusion, the reason I want to know demand before hand is because it's an API breaking change of the return values bool -> timer_id. While the meaning could be maintained (return timer_id 0 for a full timer). the type can not.

If the parameter method is unacceptable, you could instead have a global flag that you check in the callback.

I'm still convinced that by the time you have a large number of timers, maintaining the list of timer_ids to call cancel() on is the same as maintaining a list of flags to set to false. I will concede that a cancel() method is more elegant, but the added complexity is not worth it.

bogomips, 2019-05-06:
Totally right!
I must be blind, I didn't see the slot->id is exactly on the same structure that would be NULL :)
I accepted your suggestion, now the  add_task() would return NULL as the original version and 
in / at / every return the id in this way
```  return (_task) ? _task->id : 0; ```

I totally see your concerns about breaking the API.
Probably a lot of people use this lib, at the moment they would update it, their code could break and that's very annoying.
Anyhow, there is no rush, you will see if other people request such a feature.

Maybe, it could be useful if you'd add an example to show how to cancel a task by using a flag.

Of course, if you like, feel free to provide the link of my repo for those who prefer the cancel() approach.

contrem, 2019-05-07:
I've pushed an implementation with a *cancel* method to the [mc/cancel](https://github.com/contrem/arduino-timer/tree/mc/cancel) branch. It is scheduled for inclusion in the [2.0.0 release](https://github.com/contrem/arduino-timer/milestone/2). Hopefully it covers your needs.

Thanks for voicing your thoughts and guiding development towards a more usable library.

bogomips, 2019-05-09:
I've seen your code, very nice implementation! By the fact you return the whole event, a lot of code can be skipped and improved: no for loop in the cancel(), no need to deal with task_id!
I think it totally covers my needs! Great job!

It was a pleasure interacting with you, thank for the patience and the openness in trying to understand my needs/helping/implementing the code, etc.

Thanks a lot!
Pierluigi

contrem, 2019-05-10:
Great. If you decide to use it, please let me know if you have any issues.

closed 2019-12-09 by contrem