contrem/arduino-timer #5: delay value bigger than 32 768 doesn't work

nbasse, 2019-08-09:
Hi! Thank you for a great library!

When I try to use a delay grater than 32768 with the timer.in method i get weird results.
For example: `timer.in(32768, function_to_call);`

Is this some known limit of the library or some of the dependent libraries?
(Sorry if this is obvious from the code, but I´m still a beginner in c++ and don´t fully understand how this library work)
If so, is there a simple way to get around this limit?

(I´m using this on a uno where I have changed the prescalers on the internal timers to get a higher frequency PWM-output, and therefore have to multiply all time values accordingly to compensate for the timer moving faster while using functions such as millis etc. Thats why I easily get values that are quite large. 
With a compensation factor of 64, the maximum time I can have is 512ms.)

Thanks! /Nbasse

nbasse, 2019-08-09:
Ok, after some more troubleshooting with different examples i found I solution!
The above example I wrote actually works and don´t produce any weird results, but `timer.in(5122*64, function_to_call);` do.
It turns out that the compiler saves the delay as different types depending on if you type 32768 or 512*64. (maybe obvious to some, but new to me :-) )

So by adding `.0` to my multiplier and forcing it to a double (or float?) it works. <=> 
`timer.in(512*64.0, function_to_call);`

Still don't know why any of this matters, as for what I can see, it should be stored in a unsigned long.
So leaving this issue open as it might be some underlying bugs that needs to be sorted out.

contrem, 2019-08-13:
Closing as this is not an issue with the library.

Without seeing your code, it sounds like you need to use an integer suffix to make sure the compiler uses the correct type for your integer constants.

(https://en.cppreference.com/w/c/language/integer_constant)

closed 2019-08-13 by contrem