contrem/arduino-timer #6: Timer overflow

kugland, 2019-08-11:
Current implementation doesn’t account for the possibility of timer overflow, which should be a concern specially when using microsecond resolution (will happen every ~1h 12min).

contrem, 2019-08-13:
This library does account for timer overflow.

You can find the overflow/rollover tests in the [extras/tests](https://github.com/contrem/arduino-timer/tree/master/extras/tests) folder as [rollover](https://github.com/contrem/arduino-timer/tree/master/extras/tests/rollover) and [rollover-generic](https://github.com/contrem/arduino-timer/tree/master/extras/tests/rollover-generic).

Closing. Please re-open if you have found an issue.

closed 2019-08-13 by contrem

rletendu, 2020-01-06:
Sorry to reopen the discussion here, but when reading the core code at:
 line 109 :     const unsigned long duration = t - task->start;
when t is < task->start after a millis() overflow though it could trouble the rest of the code.

contrem, 2020-01-06:
That computation is by design, and correctly accounts for timer overflow.

You can find the overflow/rollover tests in the [extras/tests](https://github.com/contrem/arduino-timer/tree/master/extras/tests) folder as [rollover](https://github.com/contrem/arduino-timer/tree/master/extras/tests/rollover) and [rollover-generic](https://github.com/contrem/arduino-timer/tree/master/extras/tests/rollover-generic).

Those tests continuously overflow the timer (t < task->start == true), but the led will still blink once per second as requested.