contrem/arduino-timer #8: cannot generate timer events at "minutes"

allegfede, 2019-12-12:
Hi, i have to handle events that will occour in minutes (not milliseconds), but i noticed that using this code:

// in the global
auto play_time_timer = timer_create_default();

// in the setup
play_time_timer.in(play_time*1000, CT_win);

// in the loop
play_time_timer.tick();

// and the called procedure
bool CT_win(void *argument){
  Serial.print("timer complete!");
  return false;
}

works only if **play_time** if less than 33 seconds (so global delay less than 33000 milliseconds).

I suppose have to deal with the INT limit, but how i can make a minute counting timer?

contrem, 2019-12-13:
Make sure you're using the correct integer type for holding the values you expect to handle, and to qualify any constants with appropriate [integer suffixes](https://en.cppreference.com/w/c/language/integer_constant).

Use *unsigned long* for **play_time** and it should work. For more correctness, also qualify the **1000** constant with *UL*:

```cpp
play_time_timer.in(play_time * 1000UL, CT_win);
```

allegfede, 2019-12-13:
thanks for the reply. I'll try asap.

hartono837, 2020-01-04:
how to make function in void loop , timer.every (1000,repeat_x_times,10).
and what is opaque= NULL.thanks very much