contrem/arduino-timer #9: how do i close a timer?

allegfede, 2019-12-12:
i made a timer with:
auto delay_timer = timer_create_default();

and cannot terminate it using:
delay_timer.cancel();

timer is started with:
delay_timer.every(1000, bomb_armed);

seems he need the "task", but cannot find a way to define this task ...

contrem, 2019-12-13:
The **in / at / every** functions return the *Task*. From the README:

To **cancel** a *Task*
```cpp
auto task = timer.in(delay, function_to_call);
timer.cancel(task);
```

so:

```cpp
auto task = delay_timer.every(1000, bomb_armed);
delay_timer.cancel(task);
```

There's also a cancel example in [examples/full/full.ino](https://github.com/contrem/arduino-timer/blob/master/examples/full/full.ino#L66)

allegfede, 2019-12-13:
arduino ide replies that

task is not defined in this scope ... ore something like that.

I bypassed the problem defining Timer<1, millis> delay_timer

and running the timer with:
auto task_delay = defuse_delay_timer.every(1000, function_to_call);

results in compiles ok.

contrem, 2019-12-13:
Without seeing your code I am unable to diagnose your potential issue. Does [examples/full/full.ino](https://github.com/contrem/arduino-timer/blob/master/examples/full/full.ino) compile with your setup?

Escap3RoomsInc, 2020-01-23:
He is right though.  There is definitely an issue with cancel.  If I call cancel inside a function, I get errors.  I literally have to setup the timer again then call the cancel immediately to cancel both instances.  This has to be issue or your example is not very clear.  Using this in a real world scenario, you're not going to immediately going to call for a cancel after setting the timer like in your example.  Needs to be more clear.

contrem, 2020-01-24:
Please provide an example of the errors, and example code that isn't working properly, to help identify the issue you're experiencing.

johnmastri, 2020-02-05:
Running into the same issue - is there a way to define the task variables on the root scope without instantiating them? Otherwise, there is no way to cancel them from within another callback function call (unless I'm missing something)

```
//-- TASK VARIABLES DEFINED OUTSIDE OF SCOPE BUT NOT INSTANTIATED CAUSES AN ERROR WHEN ATTEMPTING TO INSTANTIATE WITHIN A FUNCTION
//error: 'startRoutineEvent' was not declared in this scope
 t.cancel(startRoutineEvent);

auto ledEvent;
auto countdownEvent;
auto startRoutineEvent;

void setup() {

}

void selectMode() {

  t.cancel(ledEvent);
  t.cancel(countdownEvent);
  t.cancel(startRoutineEvent);

//I WOULD LIKE TO INSTANTIATE THEM HERE
 countdownEvent = t.in(3000, showCountdown);
 startRoutineEvent = t.in(6000, startRoutine);
  
}

void cancelTimers() {

//AND BE ABLE TO CANCEL THEM HERE IF NEED BE
t.cancel(countdownEvent);
t.cancel(startRoutineEvent);

}

```

contrem, 2020-02-06:
> is there a way to define the task variables on the root scope without instantiating them?

Yes. In your example, replace:

```cpp
auto ledEvent;
auto countdownEvent;
auto startRoutineEvent;
```

With

```cpp
Timer<>::Task ledEvent;
Timer<>::Task countdownEvent;
Timer<>::Task startRoutineEvent;
```

You get an error because you can't use **auto** without an initializer.

Note that the *Timer<>::Task* constructor must match the *Timer* template if it wasn't created as the default timer.

Let me know if this helps, and thank you for providing example code and error messages.