contrem/arduino-timer #11: Multiple Arguments to callback function

NoWhere2Go19, 2020-05-02:
I was trying to handover multiple arguments to the callback function, but it did not work. However as descibed in the readme, one argument is working fine.

Therefore is it maybe possible to cast
`timer.every(10, callbackFunc_timer, color, screenBufferOld, screenBufferNew);`

to 
```
bool callbackFunc_timer(void *, uint16_t screenBufferOld[], uint16_t screenBufferNew[], uint8_t color) {
// do something
}
```

contrem, 2020-05-04:
It's not possible to cast the function signature. What you can do for multiple arguments is create a struct/class to hold the arguments and pass that object to the function.

Something like the following (untested):

```cpp
struct cb_args {
    int value_1;
    int value_2;
};

bool callbackFunc_timer(void *arg) {
    cb_args *args = (cb_args *)arg;
    int value_1 = args->value_1;
    int value_2 = args->value_2;
    delete args;
    return false;
}

void setup() {
    cb_args *args = new cb_args();
    args->value_1 = 13;
    args->value_2 = 37;
    timer.in(10, callbackFunc_timer, args);
}
```

Depending on your use of the object, you can either use a heap allocated object like in that example, a static, or a global. Do not use a stack allocated local:

```cpp
// do not do this
void setup() {
    cb_args args; // local variable
    args.value_1 = 13;
    args.value_2 = 37;
    timer.in(10, callbackFunc_timer, &args /* problem */); // bad
}
```

The local will be destroyed before the timer calls callbackFunc_timer, resulting in undefined behavior.

Hope this helps.

NoWhere2Go19, 2020-05-04:
yep, thanks for the help. I went the route of global variables. Struct would have been a good option as well, but for the dynamic memory it makes no difference. Using global variables without args with an intermediate wrapper function is working fine

closed 2020-05-04 by NoWhere2Go19