contrem/arduino-timer #12: timer.h header collides with Arduino BLE lib file (mbed os)

philj404, 2020-05-14:
Arduino IDE version 1.8.9, for Windows 10
arduino-timer library version 2.0.1.
Boards Manager support for "Arduino nRF528x Boards (Mbed OS)", version 1.1.4.

* Select "examples/arduino-timer/blink" sketch.
* Select "Arduino Leonardo" board and build.  
Build works as expected.  This is a really nice, simple and useful library.

* Select "Arduino Nano 33 BLE" board and build.

NOTE: Build fails with:

```
blink:10:14: error: 'timer_create_default' was not declared in this scope

 auto timer = timer_create_default(); // create a timer with default settings
```
However if I rename the arduino-timer library file "timer.h" to "renamedAduinoTimer.h"

  ~~#include <timer.h>~~
  #include <renamedAduinoTimer.h>

the code compiles and the timer works correctly.

I think the conflict is with the file named https://github.com/arduino/ArduinoCore-nRF528x-mbedos/blob/master/cores/arduino/mbed/drivers/Timer.h.
While Windows ignores case in filenames, with these two file names so similar there WILL be confusion for developers anyway.

Could we rename <timer.h> to <Arduino_timer.h> perhaps?
Pros:
* it's easy to tell which kind of timer we are talking about
* in the long run it's highly likely others will use the name "timer.h", too.
* it's unlikely MBED OS developers will rename Timer.h (fix this on their side) because one client wants to use the name too.

Cons:
* changing the filename breaks all legacy code.
* The Arduino folks are picky about what they attach their name to.  If they don't want to pick up this timer library, maybe we could rename it to "contrem_Timer.h" or something.

Another alternative would be to just mark this library as incompatible with Nano 33 BLE boards.  Which would be a shame because this is really a nice timer library.  

Thanks for creating this library!

contrem, 2020-05-14:
Thanks for reporting - this is definitely a concern moving forward. I'll start planning to deprecate the **timer.h** include in the next versions.

The new name under consideration is **arduino-timer.h** for the following reasons:

- It matches the library name
- Official Arduino libraries don't use this naming convention, avoiding conflicts

**Deprecation is planned as follows:**

- Phase 1 - Deprecation Warning:
  - create the new **arduino-timer.h** file alongside the **timer.h** file
  - add a **#warn** to the **timer.h** file indicating deprecation

- Phase 2 - Deprecation Error:
  - add a **#error** to **timer.h** indicating deprecation

- Phase 3 - Removal:
  - remove **timer.h**

**Please voice any concerns here.**

philj404, 2020-05-14:
Your plan makes very good sense to me.  The new name <arduino-timer.h> is quite clear and unlikely to collide with other libraries.

I would be happy to confirm your Phase 1 changes in my environment.  Just let me know when you have something to try.

IFXGAWU, 2020-06-10:
Please use a more specific name. I still I expect that <arduino-timer.h> is to generic and also many other  lib providers will may use this name. I saw inbetween some "timer.h" includes. And what do you think was the solution for them? Arduino_timer....  
Use "timer_your_3initials.h" or "timer_myspecial_name.h" or "timer_my_company.h" or whatever you want, but not a generic name everybody is intend to use.

contrem, 2020-06-13:
Changes for this issue are located in the [mc/deprecate-timer.h](https://github.com/contrem/arduino-timer/tree/mc/deprecate-timer.h) branch.

@philj404 please give that branch a test if you have time.

@IFXGAWU Sticking with the planned name for now as the work was already completed. Searching for other projects using the new name came up empty, and given that it matches this project's name, and this project is the only one named as such in the official Arduino Libraries repository, it makes sense to claim it.

philj404, 2020-06-13:
@contrem I was able to try this change with an Arduino Duemilanove and the "full" example (modified).

* The program seemed to work when I included the header "timer.h",  BUT I saw no warning for using the deprecated header name.  The "#warning" doesn't seem to do anything
* I changed the include to "arduino-timer.h", and that also seemed to work.

When I built for the "Arduino Nano 33 BLE"
* including the obsolete header "timer.h" failed (as expected)
* including the new "arduino-timer.h" header compiled.

It would be nice if the "#warning" you put in place actually worked, but at least it doesn't seem to break anything.  

I agree with @IFXGAWU that renaming the library (as well as the header) might be a good idea... but I don't know what a better name might be...  choosing a good name is hard.

So:  Your change is an improvement for "Arduino Nano 33 BLE" support.  I approve -- and thank you for making this change.  (Renaming the library is a separate discussion).

contrem, 2020-06-13:
If it's a fresh compile, the warning will be printed early, you'll probably need to scroll up:

[image]

It won't show up if warnings are disabled - but that's out of my control. Once it becomes an #error in phase 2 it will actually stop the compilation.

Thanks for testing it out.

philj404, 2020-06-13:
Aha!  Yes, my "compiler preferences" was set to "none" for some reason (I don't ever remember setting it).

When I set "compiler preferences" to "default" the #warning appeared, but didn't stop the build.  Thanks for the tip!

So... Yes, the #warning does seem to be effective.

philj404, 2020-06-26:
It looks like the arduino-timer 2.1.0 release now works with Arduino nano 33 BLE.  Thanks!

closed 2020-06-26 by philj404