contrem/arduino-timer #13: Problems with repeating tasks

iogeek, 2020-06-30:
In the example below, the timers do not repeat unless the default for add_task is changed to
    add_task(unsigned long start, unsigned long expires,
             handler_t h, T opaque, bool repeat = **1**)

#include <arduino-timer.h>

// create a timer that holds 16 tasks, with millisecond resolution,
Timer<16, millis, const char *> t_timer1;
Timer<16, millis, int> t_timer2;

bool print_message(const char *m) {
  Serial.print("print_message: ");
  Serial.println(m);
  return true; // repeat? true
}

bool pmsg(int m) {
  Serial.print("pmsg: ");
  Serial.println(m);
  return true; // repeat? true
}

void setup() {
  Serial.begin(9600);
  
  t_timer1.in(3000, print_message, "3");
  t_timer2.in(3000, pmsg, 3);
  t_timer2.in(1000, pmsg, 1);
}

void loop() {
  t_timer1.tick();
  t_timer2.tick();
  //Serial.print(".");
}

contrem, 2020-06-30:
Use the **every()** function for repeating tasks:

```cpp
void setup() {
    Serial.begin(9600);

    t_timer1.every(3000, print_message, "3");
    t_timer2.every(3000, pmsg, 3);
    t_timer2.every(1000, pmsg, 1);
}
```

**in()** and **at()** are for one time tasks.

iogeek, 2020-06-30:
thanks for the follow up.  It is what i would expect but the seeing the sample return true when called via '.in' let me to asking the intent question

closed 2020-06-30 by iogeek