contrem/arduino-timer #17: It's hard to tell if a code change is compatible across platforms

philj404, 2020-10-30:
In the past I have run into problems where my code was not compatible with a new (unknown to me) platform.  

Code changes were necessary to get it to work in new environments which I did not have access to.

It would be nice to be able to check that code changes are compatible with many Arduino variants without having to confirm by manually reconfiguring my local system for each check.

kpfleming, 2020-10-30:
On a somewhat related note, I'm currently building a Jupyter notebook which can be used to simulate any combination of timers and generate a table showing each action (at each 'tick'). Essentially this could be used to document the 'theory of operation' of the library and confirm that it matches expectations. Will probably take a few days to get it to a useful state.

philj404, 2020-10-30:
@kpfleming Yesterday I stumbled across [bxparks/UnixHostDuino](https://github.com/bxparks/UnixHostDuino) which might be able to turn Arduino simulation into unit tests.  

I think it sets up a Linux environment to run library unit tests in CI while using your Arduino sources.

Other than noticing it exists, I haven't investigated further.  First I'd like to confirm builds are successful.

contrem, 2020-11-10:
Automated cross platform builds integrated by #16 .

@kpfleming please feel free to share the work you mentioned when ready.

closed 2020-11-10 by contrem

kpfleming, 2020-11-10:
Ahh, I already did that on the other PR/issue :-)