contrem/arduino-timer #20: It would be nice to have regression tests to confirm safe code changes

philj404, 2020-11-03:
An "enhancement" request.

Sometimes when I make changes in code I accidentally break some other part of the system I didn't understand.

It can be difficult to recognize that kind of issue until after the code is released and breaks other user's builds.

It would be nice to have some regression tests which can confirm your code changes will not break other required behaviors.  Regression test results should be trivial to interpret (for example return PASS or FAIL).  There should be no question if your change broke something important.  Even if you don't understand what the problem is, you should be able to learn there is a problem.

philj404, 2020-11-04:
I was able to build unit tests and run them on my Arduino Nano.  Ok.

Now I think I got the AUnit library and UnixHostDuino working together.  This lets a timer test sketch run in a Linux build environment.  There's enough running that the test can confirm timer.at(), timer.cancel() etc. work as we would want.

So... the GitHub Action can confirm that arduino-test is still mostly-working after I make my code change.  Even when I forget to test it before a merge.

I'm going to put the automation portion of testing into a separate pull request, as the testing part is useful by itself anyway.

contrem, 2020-11-10:
Unit tests run on push integrated as of version 2.2.0 (a5ecffe87f36337c87511a1f924ce55b99c44075)

closed 2020-11-10 by contrem