contrem/arduino-timer #22: ticks() returns 0 when there are no tasks to run

philj404, 2020-11-07:
At this time ticks() returns 0 when the task queue is empty. (Bug?)

This makes it difficult to know how long to sleep before checking the queue again.  Is it time to run tick() again right now?  Is it safe to go to sleep?

I think ticks() should return ```(unsigned long) (-1L)``` to hint that the queue is idle (no tasks to run).  Then the caller can decide how how much sleep is appropriate when there are no tasks.

kpfleming, 2020-11-07:
This wouldn't be useful for me at all. I want to know how many ticks need to pass before the next timer is due.

philj404, 2020-11-07:
This is for the corner case where all pending tasks have completed.  How long can/should you sleep before you run tick() again?  
I suspect staying awake for a perpetually empty queue is going to take more power than sleeping for a while.  

I agree sleeping for ```(unsigned long) (-1L)``` (or ```ULONG_MAX``` in <limits.h>) is probably too long.  For some implementation-dependent maximum ```longest_sleep``` you should ```sleep(min(longest_sleep , ticks()))``` before waking (for example to poll for a button press or read serial input).

This makes more sense to me than making ```timer.tick() == 0``` represent two cases.  
- Sometimes ```timer.tick()``` will return 0 for "no pending tasks".  _It's safe to sleep._
- Sometimes ```timer.tick()``` will return 0 because task2 became ready while pending task1 ran.  _It's best to run ```timer.tick()``` again as soon as possible._

kpfleming, 2020-11-08:
Ahh, thanks. I hadn't considered the situation where there are no active timers at all :-)

contrem, 2020-11-09:
The way I see it *ticks()* returns the number of ticks until the next task, so if there aren't any tasks, there are no ticks until the next task.

Instead of making *ticks()* double as a check for task size, I've added *size()* and *empty()* functions in [mc/timer-size-checks](https://github.com/contrem/arduino-timer/tree/mc/timer-size-checks).

This way the user can decide how to handle a change in wait time if the timer is actually empty.

philj404, 2020-11-09:
The new timer.empty() at least lets the system figure out which case it is dealing with: "no tasks so safe to sleep" or "there are tasks ready to run".  I can get the information I need now.

Thanks for the change.

contrem, 2020-12-23:
Merged to master in ef5d1eba830ff24a0a2fb7d75fd0c7dbdc1dc4d5.

contrem, 2021-01-27:
Included in release [2.3.0](https://github.com/contrem/arduino-timer/commit/58b490866889a5a0463c7199a09db0585ccfd078)

closed 2021-01-27 by contrem