contrem/arduino-timer #24: Stops executing after some executions

GerritK, 2020-11-14:
Hey guys,

my timer stops executing after some executions. Short breakdown of my code:
```cpp
Timer<2, micros> timer;

void setup()
{
  timer.every(20000, updateLight1);
  timer.every(20500, updateLight2);
}

void loop()
{
  auto ticks = timer.tick();
  Serial.println(ticks);
}

// updateLight1 and updateLight2 reads and writes GPIOs
```

If I add `Serial.println()` to the `updateLight1`/`updateLight2` function after the gpio doing it works as expected and executes forever. If not it stops after some executions and the serial ticks output shows 0.

contrem, 2020-11-16:
Without more code / info I won't be able to help.

The following works continuously on my esp8266 using arduino-timer version 2.2.0:

```cpp
Timer<2, micros> timer;

void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);
  timer.every(20000, [](void*) -> bool {
      digitalWrite(LED_BUILTIN, 1); return true;
  });
  timer.every(20500, [](void*) -> bool {
      digitalWrite(LED_BUILTIN, 0); return true;
  });
}

void loop()
{
  timer.tick();
}
```

Some things to keep in mind:

- You're using micros, which is really fast
- There is no order guarantee for handler execution if multiple handlers are ready to run
- Hardware sometimes needs time between some GPIO actions. Adding prints might be helping your timing

philj404, 2020-11-16:
Another thing worth pointing out:  Make sure your tasks always return `true` to ensure they never stop.  

I have run into Arduino configurations (ATmega328P) which did not detect "this function does not return a value but it should have".  

This _compiles without a warning_ for ATmega328P:
```cpp
void setup() {
    Serial.begin(9600);
}

bool forgetToReturnBool() {
    int a = 0;
    a++;
}

void loop() {
    Serial.println(forgetToReturnBool());
}
```
I thought everything was fine... but my code was not always working correctly.  (What should this sketch print?  Will it ever change?)

GerritK, 2020-11-16:
Thanks @philj404 - i returned true but used void as return type. 🤦 Would have been nice if the Compiler told me that earlier. 😅 I dont understand why the Serial.println kept it alive...

closed 2020-11-16 by GerritK

philj404, 2020-11-28:
@GerritK I just discovered there's a preference setting to get stricter compiler checks!
In ```(..../Arduino15/) preferences.txt``` change
```cpp
compiler.warning_level=default
```
to
```cpp
compiler.warning_level=all
```
The good news is that this will warn you there's a problem:
```
C:\Users\philk\OneDrive\Documents\Arduino\sketch_nov15a\sketch_nov15a.ino: In function 'bool forgetToReturnBool()':
C:\Users\philk\OneDrive\Documents\Arduino\sketch_nov15a\sketch_nov15a.ino:8:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
```
The bad news is there are many libraries which also report problems -- not just your code.  So you may need to adjust warning level tolerance depending on which libraries your current sketch uses.