contrem/arduino-timer #26: "return" outside of interrupt routine

CNCQuilter, 2020-11-22:
Is it possible to designate a given timer to return false or return true outside of their respective interrupt routines?

For example, create two timers:

auto timer1 = timer_create_default();
timer1.every(500, Do_Something1);

auto timer2 = timer_create_default();
timer2.every(25, Do_Something2);

void loop()
{
  blah, blah,blah
  if (x<10)
    timer1.return false;
  if (y>0.1)
    timer2.return true;
  blah,blah,blah
}

kpfleming, 2020-11-22:
It's a bit hard to understand given the terminology you are using (there are no 'interrupt routines', these timers are co-operatively scheduled), but are you asking whether a timer's next iteration can be 'skipped', or whether a timer can be cancelled entirely?

CNCQuilter, 2020-11-22:
Kevin,

By interrupt routines, I meant DoSomething1 and DoSomething2.
return true or return false works as it should when invoked from either of these two blocks of code.
And I don't want to cancel the activity but rather only invoke a specific timer (i.e., timer1or timer2 under specified conditions). I guess skipped is the right question, but from the loop not from the DoSomething!

Thanks for your earlier super fast reply.

Dave K

CNCQuilter, 2020-11-22:
It is equivalent to something like executing the pseudocode from the loop

timer.enabled = true
timer.enabled = false

kpfleming, 2020-11-22:
Also, you are setting up two complete sets of timers here. `timer_create_default` creates a *timer set*, not a single timer. Normally you would only call that once in your program, in the `setup()` function, although you can call it outside that function if you wish.

Then, if you want to create timers, you call `.every` on that timer set object, setting an interval and a function to be called. When you call `.every`, it returns a 'handle' to that specific timer, which you can use later to cancel that timer, or reschedule it.

So, if you create the timer set properly, and create your timers properly, it is possible in the `loop()` function to reschedule an existing timer to be run later, or to cancel it entirely. Based on your example code, though, you aren't using 'elapsed time' to determine when the timer functions should be called or skipped. Please note that your timers will *never* be invoked unless you call `tick()` on the timer set object; timers do not run in the background automatically, they are completely under your control.

CNCQuilter, 2020-11-22:
It sounds like you solved my problem. I was not aware that timer_creates a timer set not just a single timer.

BTW - I am using   
 timer1.tick(); // tick timer1;
 timer2.tick(); // tick timer2;
in the loop () function, but didn't include it it my example (it was part of the blah,blah,blah).
But it was good that you re-enforced its importance by it pointing out.

Please Delete/Close the Issue.

Thanks again,
Dave K

closed 2020-12-11 by contrem