contrem/arduino-timer #37: Reuse timer

janstejskal, 2021-03-03:
How can I reuse the same timer after it's over? The timer may remain in memory and the space cannot be freed.

`#include <arduino-timer.h>

Timer<1, millis, void*> timer;
Timer<>::Task task;
uint32_t n = 0;

void setup() {
  SerialUSB.begin(115200);
  while (!SerialUSB){}
  
  task = timer.in(1000, test);
}

void loop() {
  timer.tick();
}

bool test(void *)
{
  SerialUSB.print("n: ");
  SerialUSB.println(n);

  n++;

  timer.cancel(task);
  task = timer.in(1000, test);
  
  return true;
}`

janstejskal, 2021-03-03:
```cpp
Timer<2, millis, void*> timer;
```
It works with this declaration. It is necessary to use to reserve two concurent tasks even though only one is actually needed.

philj404, 2021-03-03:
@janstejskal thanks for the source code.  It really helps me understand what the timer is trying to do.

I think ```test()``` is trying to cancel itself, which could be a problem.

- tasks should not cancel _themselves_.  The timer does it for them.
- -```timer.in(...)``` and ```timer.at(...)``` tasks only run once anyway, so no need to cancel.
- -```timer.every(...)``` tasks only need to ```return false``` to stop running.

I think your timer example needs capacity for two tasks because ```test()``` is still running when it adds a new task.  The running task is only removed _after_ it returns.

janstejskal, 2021-03-03:
> still running when it adds a new task. The running task is only removed after it returns.

Yes, that's probably how it works. I consider this a mistake proposal. A timer that has already expired should not exist. If I want to recreate the timer, I need twice the memory.

contrem, 2021-03-03:
@philj404 is correct, the timer is full since the task is still active/running and you're trying to add a 2nd task.

> If I want to recreate the timer, I need twice the memory.

In your case you should use an *every* task instead of an *in* task.

janstejskal, 2021-03-03:
No, I can't use _every_. _Every_ calls a callback at a defined interval. _In_ is defined only after the callback is completed (in my case). That is a big difference. For this reason, I did not use the library.

contrem, 2021-03-03:
Using *every* in your given example would be equivalent to using *in*, and you'd only need a timer of size 1.

If you need differing durations, then yes you would need to use *in*, and, in your given example, a timer size greater than 1. In your given example, remove the calls to *cancel*, and use a timer of at least size 2.

Memory use for this library is constant. In your given example, the timer is defined as a global, so the memory is already reserved at runtime. You will never use more or less memory, there is no possibility for a memory leak, and it can never cause an out of memory situation. Yes, it's twice the memory for a timer of size 2, but it's only an increase of sizeof(struct task), which is about 24 bytes on a uc with 4 byte longs.

kpfleming, 2021-03-03:
There is one legitimate use case for not using `every`; if you want to run the next task 1000ms *after the completion of the previous one*, you cannot use `every`. If the amount of time consumed by the task is significant, this can be a problem.

In my version of this library, I expanded the return value of the timer task to allow the task itself to indicate when it should next be run, for cases where that is important. I don't know if that is @janstejskal's issue, but it might be.

contrem, 2021-03-03:
Thanks @kpfleming.

To clarify, @janstejskal's use of *in* is perfectly valid, and provides that same functionality. He chose to reduce the timer size from its default size down to 1, and just needed to increase the timer size to at least 2.

Closing as he found a better solution for his needs.

closed 2021-03-03 by contrem