contrem/arduino-timer #38: warning: invalid conversion from 'void (*)()'

Totalis, 2021-03-08:
I am using this (brilliant) library to run several tasks repeatedly using an Arduino Uno.

I have got 3 or 4 tasks running at different intervals using:

`scheduler_timer.every(500, update_sensors)`

and similar calls, as per the example provided in the docs.

This works.  My code runs as expected and things are fine, except that there is a warning during compilation.

`Test_Rig.ino: In function 'bool send_config_data(void*)':'

'Test_Rig.ino:299:70: warning: invalid conversion from 'void (*)()' to 'Timer<4, millis>::handler_t {aka bool (*)(void*)}' [-fpermissive]
     scheduler_timer.in(250, scheduler_timer.every(500, update_sensors));'
                                                                      ^
In file included from Test_Rig.ino:15:0:
\Arduino\libraries\arduino-timer\src/arduino-timer.h:85:5: note:   initializing argument 2 of 'Timer<max_tasks, time_func, T>::Task Timer<max_tasks, time_func, T>::every(long unsigned int, Timer<max_tasks, time_func, T>::handler_t, T) [with unsigned int max_tasks = 4; long unsigned int (* time_func)() = millis; T = void*; Timer<max_tasks, time_func, T>::Task = unsigned int; Timer<max_tasks, time_func, T>::handler_t = bool (*)(void*)]'
     every(unsigned long interval, handler_t h, T opaque = T())
     ^~~~~
Test_Rig.ino:299:50: warning: invalid conversion from 'Timer<4, millis>::Task {aka unsigned int}' to 'Timer<4, millis>::handler_t {aka bool (*)(void*)}' [-fpermissive]
     scheduler_timer.in(250, scheduler_timer.every(500, update_sensors));
                             ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~
In file included from Test_Rig.ino:15:0:
\Arduino\libraries\arduino-timer\src/arduino-timer.h:70:5: note:   initializing argument 2 of 'Timer<max_tasks, time_func, T>::Task Timer<max_tasks, time_func, T>::in(long unsigned int, Timer<max_tasks, time_func, T>::handler_t, T) [with unsigned int max_tasks = 4; long unsigned int (* time_func)() = millis; T = void*; Timer<max_tasks, time_func, T>::Task = unsigned int; Timer<max_tasks, time_func, T>::handler_t = bool (*)(void*)]'
     in(unsigned long delay, handler_t h, T opaque = T())
     ^~
Sketch uses 14514 bytes (44%) of program storage space. Maximum is 32256 bytes.
Global variables use 1067 bytes (52%) of dynamic memory, leaving 981 bytes for local variables. Maximum is 2048 bytes.
`

I dont know what to do (if anything) to fix this issue.  I cannot copy too much of my code here (confidentiality and all), but I can maybe make a minimal demonstration of the problem if necessary.

Thanks

Greg

kpfleming, 2021-03-08:
Test_Rig.ino:299:70 is passing a task handle (the return value of `scheduler_timer.every()` as the handler argument of `scheduler_timer.in()`. That doesn't make any sense. This looks like the result of a copy-paste error, or something similar.

Totalis, 2021-03-08:
Yes, that also feels like a bit of a hack but it seems to work.  I basically wanted to start a repeating function such that it wont coincide with the other function (i.e. f1 running at 0 and 500ms, and f2 running at 250 and 750ms, each second). 

The documentation shows how to pass a single argument to a function but failed when I tried to pass multiple arguments.  I will de-jank this now.

kpfleming, 2021-03-08:
To pass multiple arguments to a timer handler function, you'll have to declare a `struct` to carry them, and pass the address of a instance of the `struct` as the argument (then cast it from 'void *' back to the proper type in the handler function). You'll also have to ensure that the instance lives for the entire time that the handler is in use, so probably the safest thing to do is to just make it static... but that point you might as well just have a global instance which the handler accesses by name.

contrem, 2021-03-16:
In addition to @kpfleming's response, you can also use the *Timer* constructor template parameters to set a type for the handler argument to avoid having to cast to / from `void *`.

closed 2021-03-16 by contrem