contrem/arduino-timer #46: Reset a Timer?

djusHa, 2021-04-24:
Hi,

how to clear / reset a Timer?

philj404, 2021-04-24:
**For clearing/cancelling a timer task:**
This seems to be a recurring question.  I have added discussion #47 for how to cancel a timer task.  I hope this will help.

**For "resetting" a timer task, for example to change a timer.at() time:**
I don't think that is supported directly -- you can't change the time of the next run after the task is created.  You may need to cancel the ```taskid``` for the original ```timer.at(...)``` and then call ```timer.at()``` again, to create a new task with the updated time.

philj404, 2021-04-24:
If you are resetting a "watchdog timer" task to restart the processor that seems to have crashed, I found the "avr/wdt.h" library works well (for my Arduino Uno).  See [what is watchdog timer](https://create.arduino.cc/projecthub/rafitc/what-is-watchdog-timer-fffe20).  

The watchdog timer can only do one thing: it restarts the sketch.  But it will restart the sketch even if a crash means timer.tick() never gets called.

The watchdog timer and arduino-timer libraries can both be used in the same sketch.  They do not interfere with each other.

djusHa, 2021-04-25:
Thanks! The hint with cancel the Timer and start it again was the right one.

closed 2021-04-25 by djusHa