contrem/arduino-timer #52: amount of timers

team-orion-777, 2021-08-30:
First I would like to thank you for this really easy to use and very versatile timer.

I have got a question to the amount of timers I can program. Have I got it right and the functions are limited to 16 timers? If so, is there any chance to extend this in any way, let's say 24 or 32?
As the project is growing, so is the need to control different things with the timer functions and slowly I've used all 16 ;-)

Thanks

kpfleming, 2021-08-30:
You can define `TIMER_MAX_TASKS` before including the header file,  and set it to any value you like.

team-orion-777, 2021-08-30:
> You can define `TIMER_MAX_TASKS` before including the header file, and set it to any value you like.

Thank you. I thought 'TIMER_MAX_TASKS (=16)'  in the manual is the limit of usable timers...

closed 2021-08-30 by team-orion-777

contrem, 2021-08-31:
You can also use the *Timer* constructor to decrease/increase the number of timers

```cpp
Timer<32> timer; // Timer with 32 task slots
```