contrem/arduino-timer #63: Use timerId and callbacks without argument etc.

tamtakoe, 2022-04-23:
1. Why do you use `Timer<>::Task timerTask` instead of `int timerId` like in JS? Id more understandable and you don't need pick in mind or looking for docs about `Timer<>::Task` instead of `auto` if variable is not defined.
2. Why you can't use just `void function_to_call()` without error or warnings? It is common way to return void and use no arguments in the callback. Boolean returning and arguments should be optional.
3. Create plz setInterval and setTimeout aliases for better compatibility with common standards.

skrobinson, 2022-04-25:
These questions/points all seem to come from an understanding of software timers that is strongly based on JS. Other languages and frameworks have other structures and patterns that work differently, but just as well for their use cases.

chengxg, 2022-05-04:
> 1. Why do you use `Timer<>::Task timerTask` instead of `int timerId` like in JS? Id more understandable and you don't need pick in mind or looking for docs about `Timer<>::Task` instead of `auto` if [...]

[https://github.com/chengxg/cxg-arduino-lib/tree/master/CxgJSTime](url)

closed 2022-07-04 by contrem