contrem/arduino-timer #64: Every don't work if return non true

tamtakoe, 2022-04-24:
OMG! arduinoTimer.every(500, readSerial);

philj404, 2022-04-25:
It's less reliable than that -- `timer.every()` is only _guaranteed_ to repeat when your task returns `true`. 

The function signature reserves a little stack space for a return value, but does not initialize it.  If your function does not `return` a value, whatever old garbage was left on the stack gets returned instead,  Some compliers will complain that the function should return a value, but others will not   See tip #34 in the Discussions.

kpfleming, 2022-04-25:
I've never found the logic here to be 'weird'; it's well documented that returning 'true' will cause the timer to repeat again, and returning 'false' will stop it from repeating again. Failing to actually return a value when the function declaration says it will return a value is not an issue with this library, it's a fundamental issue with writing code in C or C++, and it results in 'undefined behavior' which means the program could behave in any way at all.

closed 2022-07-04 by contrem