contrem/arduino-timer #74: Timer library affects String behaviour.

DocAlex, 2022-11-13:
Hi,

first let me thank you for this library it makes scheduling easier. 

I am using Version 2.3.1

I use the timer library to schedule sensor requests on a 328P board. (Software like it behaves as an arduino UNO).

There a stange error happens: If I set the number of timers higher than 8 - the concaternation of two strings stops working:

		lcd.print("Temperature:  "+String(SHT2X_data.Temperature));

produces no output on the display. It does not help to call:

		lcd.print("Temperature:  "+String(24));

I thinned out  the whole program it might not be compilable

If you need the complete program I can send it to you. 

Regards

Alexander

-------------------------- cut here --------------------------------------

#include "Arduino.h"
#include "Wire.h"
#include "TCA9548A.h"
#include "LiquidCrystal_I2C.h"
#include "SHT2x.h"
#include <arduino-timer.h>

	  
 uint8_t NewSensorData=1;   // True if new Sensor data to display
auto timer = timer_create_default(); // create a timer with default settings

Timer<> default_timer; // save as above
// create a timer that holds 16 tasks, with millisecond resolution,
// and a custom handler type of 'const char *
Timer<8, millis, const char *> My_timer;                            // ####################### If you set here more than 8 timers ############

//########## Timed Tasks ########
bool toggle_led(void *) {
  digitalWrite(Debug_LED, !digitalRead(Debug_LED)); // toggle the LED
  return true; // repeat? true
  }
  
  
  void DisplayPage (uint8_t PageToDisplay) {
	digitalWrite(Debug_LED2, !digitalRead(Debug_LED2));	// Humidity Sensor
	I2CMux.openChannel(AddrDisplay);
		lcd.clear();
		// Line 0
		lcd.print("SHT21");
		// Line 1
		lcd.setCursor(0,1);
		lcd.print("Temperature:  "+String(SHT2X_data.Temperature));  // ####################### This String command fails ############
		}
  
  void setup() {
	// initialize digital pin as an output.
	pinMode(Debug_LED, OUTPUT);
	pinMode(Debug_LED2, OUTPUT);
  lcd.init();
	// turn on the backlight
	lcd.backlight();
	lcd.clear();
	
	My_timer.every(500, toggle_led);
    My_timer.every(3000, HumidSensorUpdate);
  }
  
  
  void loop() {

	  My_timer.tick();
	  
	  	if (NewSensorData==1) {
		DisplayPage(PageToDisplay);
	    NewSensorData=0;
}
  --------------------------- cute here ------------------------

philj404, 2022-11-13:
You might be running out of memory for your application's variables.  The ATmega328P only has 2048 bytes of RAM.  

This RAM is used for to store variables: heap variables, stack variables and static variables. Your `I2CMux` and `lcd` objects probably need some storage, but it's not clear how much is left over for heap and stack. 

- Note that your example has two timers (`default_timer` and `My_timer`), but it only uses one of them.  You might be able to get a _little_ bit more room for more timer slots by commenting out the timer you are not using.
- If you are not careful the ATmega architecture copies string constants into RAM on startup for convenient use later.  You might be able to free up some of that wasted space by using the `F()` macro - for example replace `lcd.print("SHT21")` with `lcd.print(F("SHT21"))` to free up 6 bytes of RAM.  _When memory is tight, do this for all string constants_.  (I think this is only necessary for the ATmega "Harvard" architecture.)

philj404, 2022-11-13:
I wrote some code to print out the stack and heap limits for AVR processors with my [SimpleSerialShell ](https://github.com/philj404/SimpleSerialShell) library.  Unfortunately I think you may not have enough space to use the library as-is... but some of the code might still help you.  You may need to cut out parts to get it to work with the space available:

- https://github.com/philj404/SimpleSerialShell/blob/main/examples/ArduinoTextInterface/MemoryUsageAVR.h
- https://github.com/philj404/SimpleSerialShell/blob/main/examples/ArduinoTextInterface/MemoryUsageAVR.cpp

philj404, 2022-11-13:
Here is some code which can print the contents of memory (RAM).  For only 2048 bytes, it can be VERY helpful for finding wasted RAM which only contains string constants.  Unfortunately it is set up for use as shell command.  You will need to modify it so it works for you:
- https://github.com/philj404/SimpleSerialShell/blob/main/examples/ArduinoTextInterface/memoryCommands.cpp

DocAlex, 2022-11-13:
> You might be running out of memory for your application's variables. The ATmega328P only has 2048 bytes of RAM.

Thank you very much for the quick help. 

I removed the unused timers an now I was able to create 35 timers. Thus I assume you are right in your guess that there is a memory problem. 

Maybe a hint to this behaviour in the documentation might be a good idea.

Thank you again for your help. 

Alexander

philj404, 2022-11-14:
The "running out of RAM" issue is more tied to the older ATmega architecture than it is to <arduino-timer.h>.

If you need to use Arduino UNO and related processors, _and_ you are running out of RAM space, it may help to learn about `__FlashStringHelper` and other tricks to reduce RAM usage.

If you haven't committed to ATmega, newer processors such as ESP32, STM32 etc. have much more RAM and make this less of a concern.

closed 2023-02-05 by contrem