contrem/arduino-timer #77: Bug: Behavior Inconsistency when canceling a timer

adelin-mcbsoft, 2023-02-02:
As the title says, there is a bug / inconsistency when canceling a timer.
Let's say we have a timer: `Timer<10> MainTimer; // Timer with 10 task slots` .

If we assign a Task ID to a certain task, e.g. `Timer<>::Task DummyTask;` and set a task for it, e.g. `DummyTask = MainTimer.every(1000, DummyFunction);`, the library behaves differently when canceling the timer from the inside (by returning `false` into the `DummyFunction`), compared to canceling it from the outside (by calling the `MainTimer.cancel(DummyTask)` method ).

When canceling the timer using the `.cancel()` method, `DummyTask` gets back ID (value) = 0, as it should.
But when canceling the timer from the inside (by returning `false` in `DummyFunction`), `DummyTask` retains its original ID, without reverting back to `0` as it should.

I created two demo examples here:
This one demonstrates the good behavior (using the `cancel()` method): https://wokwi.com/projects/355520237400512513
And the other one, demonstrates the wrong behavior (by retaining its task ID whenever using `return false`): https://wokwi.com/projects/355520290434348033 .

I'm not an experienced C++ developer, but I suppose the code misses a task reference when removing it, [here](https://github.com/contrem/arduino-timer/blob/master/src/arduino-timer.h#L137).

(I'm wondering if this could also explain the fact that I found situations when the timer cancels itself, as per my other opened issue, #76 ).

Thanks,
A.

philj404, 2023-02-03:
First, let me say the examples are nice and easy to understand. The wiki helps make it VERY clear what it is doing.

I think this is "expected behavior" -- even though I did not expect it myself.

There is a difference between `timer.cancel(aTaskId);` and just a `return false` for a task which should not run again.  See the source for cancel:
```
    ...
    /* Cancel the timer task */
    void
    cancel(Task &task)
    {
        if (!task) return;

        timer_foreach_task(t) {
            if (t->handler && task_id(t) == task) {
                remove(t);
                break;
            }
        }

        task = static_cast<Task>(NULL);
    }
    ...
```
Note that `cancel(task);` nulls out the Task passed in (this is the part I had not realized).  This is a nice courtesy to make it more obvious the saved Task is obsolete.  Note that other copies of the Task (if any) will _not_ get nulled out.  (I'm not sure why you would want more than one copy of the ID, but if you did, it will be stale.)

But a task canceling _itself_ with `return false;` does not null out references to its ID -- it doesn't even know if there are any saved.

adelin-mcbsoft, 2023-02-03:
Yes, that's my main point... that it doesn't know and I think it should.

In the end... what's the point of keeping the task ID ( > 0) in memory after the task is canceled? It's no useful info.
However, having it `0` could be very useful, as it confirms that task is no longer active (and also, "equalizes" to the `.cancel()` behavior).

philj404, 2023-02-04:
The task and its task ID _inside the Timer_ is zeroed out -- but you kept a copy of the Task when you created it.  The Timer does not keep track of how many copies you made (if any).

However, `timer.delete(obsoleteTask)` can zero out the `obsoleteTask` that was passed in -- the only one it knows about.  If you made other copies, they are still not zeroed out.

adelin-mcbsoft, 2023-02-04:
_I suppose you meant `timer.cancel(obsoleteTask)` instead of `delete` (as there's no such method in the code)._

Anyway, now I understand where you're coming from and why is it behaving like this.
But unfortunately, this leaves no other way of tracking if a task is still running or not, without additional code in the callback function. (in each callback function, to be precise... which is bad).

That's why I was expecting that the return value of the `timer.every()` / `timer.in()` would be a (global) reference/pointer in memory to the task ID, that would also get zero'ed when the timer is `remove()`'d.

philj404, 2023-02-04:
Oops  🙄 .  Yes, I really meant `timer.cancel(obsoleteTask)`.  I'm going to edit my comment to make more sense to other, but thanks for pointing that out.

The trouble with returning a pointer the (internal) task ID instead of a copy is that the slot will get reused at some time when another task is added.  Then a `timer.cancel(...)` call using the stale reference would cancel... something else.

adelin-mcbsoft, 2023-02-04:
Then... which approach would you think is best to track if a timer is still running or not, without any additional code in the callback?

_(1.)_ Although I'm starting to consider zero'ing the task ID just before returning `false`.
_(2.)_ Calling `timer.cancel()` from inside the function (before or instead returning `false`) also came into my mind, but I find this solution a bit ugly... I'm not sure it was designed to work like this.

philj404, 2023-02-04:
Do not call `timer.cancel()` from inside the task as it is running as that won't work -- see #47.  

Setting the saved `Task` to `NULL` before returning `false` should work (if casting `savedTask = (Task) NULL;` works).  Then any time the saved `Task` is not `NULL` the task must be active.  Make sure that there's exactly ONE saved copy to maintain and you're probably OK.

adelin-mcbsoft, 2023-02-07:
Given the upcoming changes in the future ([next](https://github.com/contrem/arduino-timer/blob/next/src/arduino-timer.h)) version and the fact that the task ID is now a reference to the pointer and no longer an ID, is it still not recommended to call `.cancel(savedTaskReference)`  in an `.every()` loop function, **before** `return false` ?

My goal is to null-out a (global) `savedTaskReference` before the timer cancels itself with the `return false`.
Is much more _cosmetic_ to do a 
```
timer.cancel(savedTaskReference)
return false
```

than

```
savedTaskReference = static_cast<Timer<>::Task>(NULL);
return false
```

From my testing, they do the same thing _(but the `cancel()` method has some overhead -- indeed).
The main difference is that in the first code-block the timer is canceled at `line 1` (with the `cancel()` method), while in the second code-block the task is canceled at `line 2` (`return false`).

I'm not aware if doing this with the `cancel()` method corrupts the memory or anything, as it is executed before the `every()` function finishes execution ... hence my question.
Thanks in advance!

philj404, 2023-02-07:
Related topics: #47 , #73.

Recap: At this point repeating tasks should just return `false` rather than cancelling themselves.

As I remember, if you cancel a repeating task from inside itself instead of returning `false`, you have to be very careful about your task not adding any tasks after the cancel.  Otherwise there is a good chance the dying task will remove the new task from its own obsolete slot on when it cleans up/exits.

It gets complicated for the timer to recognize if the active slot's contents were replaced on exit and NOT delete them during cleanup.

adelin-mcbsoft, 2023-02-07:
Got it.
Better safe than sorry, so I'll just cast the reference to null (`savedTaskReference = static_cast<Timer<>::Task>(NULL);`) before returning `false` and that's it.

Thank you for all the effort you've put to give me the proper information.
I'm gonna leave this issue open few more days just in case anybody has anything else to add, and I'm gonna close it afterwards.
Many thanks once again!

closed 2023-02-09 by adelin-mcbsoft