contrem/arduino-timer #86: reseting the timer

dave5992, 2023-08-28:
It would be great if we could get a reset funtion

philj404, 2023-09-01:
I'm not sure what you mean by a 'reset function'.  

Does `timer.cancel(aTaskId);` do what you want?

dave5992, 2023-12-09:
if timer.cancel(); resets the time to 0 then yes.

philj404, 2023-12-09:
By default the timer uses `millis()` to determine the current time (time since last hard reset).  The timer only reads the value of `millis()` -- never modifies it directly.  I don't know if there's a way to reset it to zero without a hard reset.

If `millis()` starts returning earlier values, the timer will get confused about when a pending task should come due.

`timer.cancel()` just clears all pending tasks.  That's probably good enough to do what you need.  The timer tasks don't particularly care what value `millis()` returns, they just check if `millis()` has _increased_ beyond the delay requested (relative to when the task was created).

(This is probably not worth the effort, but...)
You can substitute your own function (and do your own resetting) instead of using the default `millis()`.  See an example simMillis() in  https://github.com/contrem/arduino-timer/blob/master/extras/tests/timerTest/timerTest.ino.  I use it to run tests without actually waiting for `millis()` to elapse/increment over seconds.

closed 2024-10-17 by contrem